G4lile0/ESP32-WiFi-Hash-Monster #1: Recipe for Odroid-GO

tobozo, 2019-11-18:
This is a great project, I love it :heart_eyes: 

As the maintainer of the ESP32-Chimera-Core I can't help but see every M5Stack application as a potential Odroid-GO application, so I tried to get it working outside the scope of your ReadME and somehow got successful.

I haven't checked yet if the capture to the SD is really operative, but the UI loads successfully is very similar to what I get on the M5Stack.

[image]

It worked by using the alternate [ESP32-Chimera-Core](https://github.com/tobozo/ESP32-Chimera-Core) (100% compatible M5Stack core) and adjusting bits of code (e.g. disabling the FastLed) and selecting "Odroid Go" in the boards menu.

Does it demonstate a working scenario with PSRAM enabled?

It's also good to know your project is compatible with the [SD-Updater](https://github.com/tobozo/M5Stack-SD-Updater).

Odroid-GO Turing path:
  - M5.begin() replaced Serial.begin() as the first statement in the setup() => much less strange bugs when booting
  - SD.begin( TFCARD_CS_PIN ) turns out to be a better multi-board standard than maintaining SPI manually
  - Inheriting from existing macros can prevent pin conflicts:

        #define BUTTON_PIN_A BUTTON_A_PIN // button to change the channel
        #define BUTTON_PIN_C BUTTON_C_PIN // button to change the bright of the LCD

  - Got rid of the rotation (not needed by M5Stack or Odroid-Go anyway)

        // M5.Lcd.setRotation(1); // already performed by M5.begin()

  - Used the syntax of the {M5Stack|Chimera} core

        M5.Speaker.write(0); // Speaker OFF
        // dacWrite(25, 0); // Speaker OFF

  - Isolated all FastLed statements, including not launching the blinker task:

        #ifndef ARDUINO_ODROID_ESP32
        #include <FastLED.h>
        #endif

Do you want a PR for this ?

tobozo, 2019-11-18:
[update]

  - All .pcap files created on the SD are empty, will investigate
  - Main benefit of using an Odroid-GO is to make proper screenshots (TFT_MISO is wired):

[image]

[image]

[image]

[image]

[image]

luminouw, 2020-07-09:
Hello,

I just stumbled upon this project and also have an Odroid-GO at home to play with, should I just follow your steps to make it work ? : )

Thanks

tobozo, 2020-07-09:
@luminouw I haven't tried with the latest version but I'll be happy to hear your feedback and provide help.

the changes that occured since this topic was created is that now the ESP32-Chimera-Core is using the LovyanGFX display library as a separate dependency (available from the library manager), so you'll have to make sure you have the latest versions of those libraries installed, and that you selected the Odroid GO from the Arduino boards menu when you compile so that the Odroid-Go device is properly configured at compilation time.

possible glitches to watch : the purple monster looks scrambled -> can be easily fixed by commenting out one of the `setSwapBytes` statements (can't remember which one though)

please let me know how it went :-)

shojkeee, 2020-11-11:
Constantly says "insert sd" what to do?

tobozo, 2020-11-11:
bad sdcard ?

shojkeee, 2020-11-11:
sorry. After I wrote a comment, I installed sd 16 gb and everything worked. I thought 2 gb was necessary

SaiBot-YT, 2020-11-12:
can you please add some more information about the SD Card? i was using a 32Gig Sandisk Ultra, and it doesnt work. also formatting it to 16GB of Fat32... I dont know hot to fix the issue "INSERT SD"

tobozo, 2020-11-12:
1) make sure you've selected "ODROID_GO" from the arduino boards menu
2) use a smaller partition on the SD card (e.g. 4GB)
3) check your library manager to see if the libraries ESP32-Chimera-Core and M5StackUpdater are up to date
4) disable psram

[image]

shojkeee, 2020-11-12:
I have a 16 Gb card divided into 2 partitions, one of the partitions is 2 GB.