G4lile0/ESP32-WiFi-Hash-Monster #4: Interested in new features/fixes?

scriptguru, 2020-04-09:
I've made a few fixes/improvements locally and they work well. Should I create a PR?

Fix: changed MAX_SSIDs from 256 to 1792
The limit of 256 access points was very easily achievable if the device was moved (like when you walk with it). I've tried 2048 but it doesn't fit in RAM, so I've settled at 1792, it should be enough for a pretty long walk.

Improvement: now it's possible to make it cycle through pre-defined channels automatically. In North America it makes sense to cycle through channels 1, 6 and 11 only, because those are the only non-overlapping channels in 2.4GHz range, and that's where most APs are. The interval of the channel auto-switching and the channels are configurable.

Improvement: SD card is on by default (configurable, of course).

Also made a few minor changes related to calls to draw() function in the main loop.

Also there is a bunch of commented lines in the code and they probably should be removed, but I haven't touched them because I thought they might have been left there for a reason.

G4lile0, 2020-04-09:
Yep sure, thanks for your work, send me the PR 👍I'll take a look on the commented code.. Thanks and stay safe!!

scriptguru, 2020-04-10:
Great!
I've created the PR.

G4lile0, 2020-04-10:
Thanks Val, 
I like the idea of cycling through channels, is useful for Wardriving, but if you want to "audit" an specific wifi network is better to keep on that channel...  let me test it.  Thanks!

G4lile0, 2020-04-11:
PR, accepted :+1: Thanks!
Soon I'll add the option to enable/disable auto-switching from the device.

closed 2020-04-11 by G4lile0