G4lile0/ESP32-WiFi-Hash-Monster #6: Non-empty .pcap files converted to empty .hccapx

scriptguru, 2020-04-10:
I have many non-empty .pcap files created by the ESP32 WiFi Hash Monster, but most of them have been converted to empty .hccapx files. Also I've read that .hccapx is "specifically designed and used for hash type -m 2500 = WPA/WPA2".

However, as I understand, the Hash Monster also captures PMKID packets, which are hash type 16800 (see [https://hashcat.net/forum/thread-7717.html](https://hashcat.net/forum/thread-7717.html) for more details).

To me it looks like the tool that is recommended for .pcap conversion is simply ignoring everything that is not a traditional 4-way handshake.

Any thoughts how to extract PMKIDs from the .pcap files?
I haven't searched for such a tool yet, but perhaps you know what tool to use already and just forgot to update the docs.

A side note - Pwnagotchi also stores .pcap files and as I understand it writes packets that contain PMKID to them

G4lile0, 2020-04-10:
Long time since I programmed it  but I think that I store on the SD some non (PMKIDs /  4-way handshake)  that are need it for knowing the SSD or other stuff (You can check it, with wireshark). 

For 4-way handshak I use cap2hccapx to convert from **pcap** to **hccapx** then I use hashcat to calculate the original key from the hash.

For PMKID, hcxpcaptool is the right tool, here you have a guide, hope it helps:

https://www.shellvoide.com/wifi/hashcat-guide-how-to-brute-force-crack-wpa-wpa2/

RobRusso, 2020-07-28:
I only received empty pcap files or invalid handshakes Whats can be wrong?