G4lile0/ESP32-WiFi-Hash-Monster #7: Gps support and on demand deauth

bking46, 2020-05-07:
Hi All,

I am quite new to the m5stack platform and wanted to ask for your support on this please. I am enjoying this device a lot and wanted to check whether there is a possibility to add gps support or on demand deauth? For the on demand deauth I was thinking of for example on using the long press on the middle button to deauth a network if that would be possible? Many thanks for sharing this project, I am enjoying it quite a lot.

Apologies for posting this in the issue section. This is not an issue, it is more of a functionality request.

tobozo, 2020-05-07:
ESP32 can't do deauth (only sniffing) so basically you'd need a dedicated ESP8266 to do the deauth job for you, quite easy to setup with a regular M5Stack proto board but possibly problematic (e.g. brownout) when both WiFi interfaces are enabled

[image]

bking46, 2020-05-07:
Thanks so much for the prompt response. Is the picture you are showing a deauther on an m5stack pronto? Is the deauth done from software or from one of those two buttons? Looks very compact and would definitely be interested to try if it works. Also how do I revert back the m5stack fire to have uiflow? Just got the m5stack fire today and the first project to upload was the  hash monster😁

tobozo, 2020-05-07:
The recent version of the deauther has a serial command interface, so it's technically possible to use a button on the M5Stack to start a scan + deauth on the ESP12, but using this shield with @G4lile0's WiFiHash Monster gave me a lot of packet loss and brownout errors, maybe the 3d antenna on the M5 is very ticklish, or maybe using the serial while capturing packets is a bad idea ?

[M5Stack-ESP12-Shield](https://github.com/tobozo/M5Stack-ESP12-Shield)

bking46, 2020-05-08:
Thanks so much for sharing this with me. Would you happen to know if I can use the Lovyan Launcher to run Hash Monster?

tobozo, 2020-05-08:
if it's compiled with m5stack-sd-updater, it works with the lovyan launcher

zenonasz, 2020-05-15:
> ESP32 can't do deauth (only sniffing) so basically you'd need a dedicated ESP8266 to do the deauth job for you, quite easy to setup with a regular M5Stack proto board but possibly problematic (e.g. brownout) when both WiFi interfaces are enabled
> 
> [image]

i'm confused?!
https://github.com/GANESH-ICMC/esp32-deauther

tobozo, 2020-05-15:
oh it finally happened, thanks for pointing this out !

bking46, 2020-05-16:
> if it's compiled with m5stack-sd-updater, it works with the lovyan launcher

Thanks so much, have tested and it works perfectly with the launcher.

zenonasz, 2020-05-16:
> oh it finally happened, thanks for pointing this out !
> 
> [edit] doesn't look like it can be used out of the box with the WiFi-Hash-Monster

 esp-idf sdk though, not sure if it can be ported but still there a way of doing it

bonedaddy, 2020-12-02:
A short term solution is to use a pwnagotchi in combination with the hash monster, seems to be working "okay" for now.

tobozo, 2020-12-02:
there's an ongoing discussion about some hybrid ESP32/ESP8266 build on [ESPMarauder](https://gitter.im/justcallmekoko/ESP32Marauder)'s gitter

pwnagotchi is neat, but it constantly tries to call home and enroll, even with grid disabled

I stopped playing with that after it blew one of my micro sd cards by filling up the logs with useless messages

[image]

tonysoprano187, 2021-01-13:
@tobozo any advice in trying to instal the deauth via Arduino? many thanks :)

RChadwick7, 2021-01-19:
Is adding an ESP8266 an implemented method of deauth, or more of a theoretical thing? Did you try putting caps across 3.3V and Gnd at the ESP8266?

tobozo, 2021-05-27:
This article highlights two new ways to achieve deauth on ESP32, the Rogue AP solution seems to be the best choice for Arduino as it doesn't need to be maintained across sdk's. 

https://hackaday.com/2021/05/27/wifi-penetration-testing-with-an-esp32/

Has anyone stumbled across an Arduino POC using the Rogue AP solution? Found [this](https://github.com/Mr-Blonde/RogueAP) or [this](https://github.com/TheCamp/esp32-rogue) but found no traces of `esp_wifi_deauth_sta()` so far.