G4lile0/ESP32-WiFi-Hash-Monster #8: Which unit?

pingywon, 2020-09-08:
I am confused about if this works on:

https://m5stack.com/collections/m5-core/products/grey-development-core?variant=16804796006490

or:

https://m5stack.com/collections/m5-core/products/fire-iot-development-kit

The pictures look like Fire, but it isnt stated anywhere.

tobozo, 2020-09-08:
it should work on M5Classic too

pingywon, 2020-09-08:
Soooo the fire than?

G4lile0, 2020-09-08:
The Fire IoT is perfect, Hash monter will use the RGB LED bars to show deauth or hash packet even with the screen off, but it will work with the other version of the M5Stack.

pingywon, 2020-09-09:
Thank you for a clear and concise answer.

kbwrx, 2020-09-10:
Will the Hash Monster also work on the M5Stack Core2 ESP32 IoT Development Kit (https://m5stack.com/collections/m5-core/products/m5stack-core2-esp32-iot-development-kit?variant=35960244109476)? Every time I have tried uploading the sketch, I get nothing. Just a blank screen and I cant power on the unit. I verified the sketch prior to upload. I was also able to upload other examples to verify it works. Thanks for any help!!

tobozo, 2020-09-11:
@kbwrx got it working on my M5Core2 but a few changes are required to get it working

[image]

@G4lile0 I'll make a PR as soon as I can, but I'm wondering if something could be done with the haptic interface :thinking:

kbwrx, 2020-09-11:
@tobozo Thank you sir! I look forward to the PR.

tobozo, 2020-09-12:
This procedure is very experimental, works today and probably will be broken tomorrow, so here's the current recipe :

1. Add the [M5Stack boards](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/arduino/package_m5stack_index.json) to the Arduino IDE boards list and open the Boards Manager 

[image]

2. Do not install M5Core2 library from the library manager, use [this fork](https://github.com/lovyan03/M5Core2) instead

3. Make sure you've upgraded to [M5Stack-SD-Updater v1.0.0-alpha](https://github.com/tobozo/M5Stack-SD-Updater/releases/tag/1.0.0-alpha) it is already available from the Arduino Library Manager

4. Checkout the code from [this fork of ESP32-WiFi-Hash-Monster](https://github.com/tobozo/ESP32-WiFi-Hash-Monster/)

5. Before flashing, select "M5Core2" from the "M5Stack Arduino" boards menu

[image]

Notes: all of this is temporary, a [PR is in progress](https://github.com/m5stack/M5Core2/pull/9) for M5Core2 Touch/Buttons emulation, and I can't make a pull request to the WiFi Hash Monster until this has landed in the official release.

G4lile0, 2020-09-12:
I doesn't have M5Stack-Core2, thanks a lot for your work @tobozo  !!

kbwrx, 2020-09-12:
@tobozo I got it working with your guide! Thank you for your help and your time! Ill keep an eye out for future PRs

nimmen, 2020-09-17:
> it should work on M5Classic too

Does not, tried flashing it through arduino IDE, compiles fine, then as I'm uploading it, device just starts rebooting constantly until arduino throws "A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header"

arduino 1.8.13
M5Stack-SD-Updater v1.0.0-alpha(had to add it as did not compile)
M5Stack-0.3.0
ESP32-1.0.4 board

omdv, 2020-09-17:
@tobozo, thanks, your guide worked with one correction.

Looks like there is work and several PRs going to have backwards compatibility of buttons, which is the main differences from Core to Core 2. I ended up using https://github.com/ropg/M5Core2 fork, instead of yours. I think eventually the PRs to support buttons will penetrate to m5core2 library directly.

nimmen, 2020-09-20:
> > it should work on M5Classic too
> 
> Does not, tried flashing it through arduino IDE, compiles fine, then as I'm uploading it, device just starts rebooting constantly until arduino throws "A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header"
> 
> Arduino: 1.8.13 (Windows 10), Board: "M5Stack-Core-ESP32, QIO, 40MHz, Default, 921600, Info"
> M5Stack-SD-Updater v1.0.0-alpha(had to add it as did not compile)
> M5Stack-0.3.0
> ESP32-1.0.4 board

Solved this by updating CP210x USB drivers, so nothing to do with codebase. Works fine with M5Stack-Core-ESP32

closed 2020-09-24 by pingywon

Calevra1, 2024-02-11:
I have m5stack core metal and cant make it work with the hash monster