G4lile0/ESP32-WiFi-Hash-Monster #9: Power consumption

cookins, 2020-09-10:
Hello.
I'm sorry about question, it's not exactly about firmware, at least partly. So on this firmware, what average power consumption on m5stack core unit? Or at least tell me for how long will it run? They selling with additional battery, so I'm curious what will be difference. Thanks and sorry once more.

G4lile0, 2020-09-12:
Using the "incognito mode" will disable the display and the led bars to reduce the consumption, so it will run for a few hours, if you plan to run for long periods I suggest you to use an external power bank.

RChadwick7, 2020-09-16:
I measured about 100mA with the display off, and about 120mA with it on.

cookins, 2020-09-17:
> I measured about 100mA with the display off, and about 120mA with it on.

Really? Even esp8266's consumption was 215 mAh, while transmitting data at full power. It's cool, thanks.

closed 2020-09-17 by cookins