G4lile0/ESP32-WiFi-Hash-Monster #16: Adding an off switch

RChadwick7, 2020-10-12:
Can the power button be programmed to shut down?

tobozo, 2020-10-12:
Short answer : Yes, the power button already acts as a shut down, but no it can't be programmed.

A double push on the reset/power button shuts the M5 down.

However when the M5 is plugged to the USB the power module is immediately turning it back on, and there's nothing you can do about that unless you use another event to trigger a M5.PowerOff() 

If you want to see that software PowerOff in effect, just remove the SD Card and reset the M5, then wait 60 seconds and it'll turn off by itself.

RChadwick7, 2021-03-24:
Do you know if the power button was implemented somewhat recently? I never got the double-push to shut down (The first push of the button just reset it, as well as the second push). I shut down by pulling the SD, resetting, and waiting for it to eventually shut down. If I get a chance to recompile, I'd probably shorten the delay

toto99303, 2024-01-18:
Same here, double click does not work.