G4lile0/ESP32-WiFi-Hash-Monster #18: No captures

toglader, 2020-11-13:
My ESP32 WiFi Hash Monster doesn't capture anything to the SD card. All pcap-files are empty. It has been running now for a week but it hasn't been able to capture any PMKID packet. What could be wrong? I have tested Automatic, Smart and Manual Channel mode.

tobozo, 2020-11-13:
Hello and thanks for your feedback.

I just made a 3mn test on an Odroid-Go and the pcap files are nonempty, what device are you using ?

Supported devices are : M5Stack Classic/Fire/Core2 and Odroid-Go

The pcap files aren't saved on the root of the sdcard any more, if you just updated your firmware you should now check in the "/pcap" folder instead, and yes it may eventually contain *some* empty pcap files.

Also how many WiFi devices does your smartphone show when you scan your nearby WiFi network ? If everything around you is using 5GHz WiFi it's no surprise your pcap files stay empty, ESP32 is limited to 2.4GHz.

Now it could also be a SD Card problem, setting "Core Debug Level" to "Warn" in the "Tools" menu before flashing may eventually show you more errors/warnings.

Some tests you can eventually perform: 

  - Use a 4GB partition
  - Use another SD Card

toglader, 2020-11-13:
I'm using M5Stack Basic Core. Isn't it supported?
There isn't any problem with sdcard access. M5 is able to write to card (/pcap folder exists). Problem is that all pcap files are 0 bytes long. Seems like total hash value stays in zero in M5stack display.

Here is many 2.4GHz Wifi networks present, so the problem isn't in frequency band.

tobozo, 2020-11-13:
Thanks for that information, trying to reproduce on my M5 Classic (basic core) but works for me.

All I can think of is the library versions you're using, maybe see if LovyanGFX and/or ESP32-Chimera-Core can be updated from your arduino library manager ?

More on debugging if your libraries are already up-to-date:

Does the serial output emit any extra information when Core Debug Leve is set to warn ?

You should at least see that bit of output when packets are captured for a while (a minute is enough).

[image]

I suggest switching immediately to automatic mode for a faster result.
Whenever you see a message hinting at a file name + file size, turn the SD Card off (long press on BtnB) and take it off, then check that the file size is the same.

If you don't see any of those messages, then no packet has been captured and the SD isn't the problem, better check for the esp32 SDK version (arduino boards manager), the state of the 3d antenna, or signal interferences (e.g. coming from the USB cable or nearby WiFi devices).

SaiBot-YT, 2020-11-13:
I have the same Problem, but there is no way i can chage it... Here are my files (most of them are 0mb too)

is there a way of getting the files / contents via 

[pcap.zip](https://github.com/G4lile0/ESP32-WiFi-Hash-Monster/files/5537188/pcap.zip)

tobozo, 2020-11-13:
@SaiBot-YT none of those pcap files seem to have been named by ESP32-WiFi-Hash-Monster, moreover the creation date/time do not seem to match the ESP32 defaults (1980 instead of 1970), are you using the latest build ?

tobozo, 2020-11-13:
Assuming you already did all the pre-checks listed in this thread and you're sharing the same config as @toglader , then it's the same debugging method: enable core debug level=warn when flashing, keep the serial window open and look for sd card i/o errors or warnings in the output

SaiBot-YT, 2020-11-13:
i did, and it didnt fix the issue... but after formatting the sd card with cmd and allowing all permissions it works now... semms sus

SaiBot-YT, 2020-11-13:
can you add the option to format the card automatically if it is broken or what so ever

tobozo, 2020-11-13:
well I'm not the repository owner and have no idea how to format a SD Card from an ESP32 but I'm sure that if you come up with a solution to that problem and make a pull request @G4lile0 will be happy to review your code :eyes:

SaiBot-YT, 2020-11-13:
ok. ill try it

toglader, 2020-11-19:
> If you don't see any of those messages, then no packet has been captured and the SD isn't the problem, better check for the esp32 SDK version (arduino boards manager), the state of the 3d antenna, or signal interferences (e.g. coming from the USB cable or nearby WiFi devices).

I didn't see any of those messages during testing. However there is about 5 0 bytes long pcap files in sd card.
Display shows for zero total deauths and one total ap hash. Hashes in total 7. 

In serial console there isn't anything except auto-switching channels messages and warining about non-ascii characteres:

[W][ESP32-WiFi-Hash-Monster.ino:637] wifi_promiscuous(): NO IMPRI 00 - 20

[W][ESP32-WiFi-Hash-Monster.ino:637] wifi_promiscuous(): NO IMPRI 03 - 20

[W][ESP32-WiFi-Hash-Monster.ino:641] wifi_promiscuous(): NO ASCII 03 - 20

[W][ESP32-WiFi-Hash-Monster.ino:637] wifi_promiscuous(): NO IMPRI 04 - 20

[W][ESP32-WiFi-Hash-Monster.ino:641] wifi_promiscuous(): NO ASCII 04 - 20

Esp32 version in boards manager is 1.0.4.

shanbady, 2021-05-28:
@SaiBot-YT can you elaborate on what fixed it for you. im having the exact same issue with a 32g sd card