G4lile0/ESP32-WiFi-Hash-Monster #20: Error Class M5Display Has No Member...

bonedaddy, 2020-11-30:
I'm attempting to use this project with an m5stack-core and am unable to build the code. I receive  numerous errors about class M5Display not containing various members. 

```
.pio/libdeps/m5stack-core-esp32/M5Stack-SD-Updater/src/M5StackUpdaterUI.h:110:36: error: 'class M5Display' has no member named 'textsize'
     SDUTextStyle.textsize    = tft.textsize;
                                    ^
.pio/libdeps/m5stack-core-esp32/M5Stack-SD-Updater/src/M5StackUpdaterUI.h:111:36: error: 'class M5Display' has no member named 'textdatum'
     SDUTextStyle.textdatum   = tft.textdatum;
                                    ^
.pio/libdeps/m5stack-core-esp32/M5Stack-SD-Updater/src/M5StackUpdaterUI.h:112:36: error: 'class M5Display' has no member named 'textcolor'
     SDUTextStyle.textcolor   = tft.textcolor;
                                    ^
.pio/libdeps/m5stack-core-esp32/M5Stack-SD-Updater/src/M5StackUpdaterUI.h:113:36: error: 'class M5Display' has no member named 'textbgcolor'
```

tobozo, 2020-11-30:
hey @bonedaddy please make sure you're using the latest version of the following libraries (from the library manager):

- LovyanGFX
- ESP32-Chimera-Core
- M5StackUpdater

bonedaddy, 2020-12-01:
I double checked and I'm still getting the issue. I'm attempting to use PlatformIO through VSCODE so I wonder if that might be the issue?

tobozo, 2020-12-01:
[Platformio library page](https://platformio.org/lib/show/5523/ESP32-Chimera-Core) shows a version number that's not even listed in the official releases, my guess is they probably checked out from the master instead of picking up a release.

The current version should be `tobozo/ESP32-Chimera-Core@1.1.3`.
You may try to update the version number in your platformio.ini or just get set it manually using one of those:

  - `pio lib install tobozo/ESP32-Chimera-Core@1.1.3` <<< will probably fail
  - `pio lib install https://github.com/tobozo/ESP32-Chimera-Core/archive/1.1.3.zip`

There's a high probability other library versions are affected too, here's what I'm currently using:

LovyanGFX @ 0.2.7
ESP32-Chimera-Core @ 1.1.3
M5Stack-SD-Updater @ 1.0.2

bonedaddy, 2020-12-01:
That solved the issue but now another one is happening

```
.pio/libdeps/m5stack-core-esp32/M5Stack-SD-Updater/src/M5StackUpdater.h:167:18: fatal error: SD.h: No such file or directory
```

I ran `pio lib install https://github.com/tobozo/M5Stack-SD-Updater/archive/1.0.2.zip` to make sure I was using the same version as you beforehand.

tobozo, 2020-12-01:
can't remember why this happens but I know such platformio-specific issues are usually solved by including the "missing but installed" library from the top of the .ino file:

    #include <SD.h>

bonedaddy, 2020-12-01:
tried but no luck :( let me play around with this a bit to see if i can figure it out

tobozo, 2020-12-01:
oh found the curlpit in the platformio.ini

```ini
;lib_ldf_mode = chain+
lib_ldf_mode = deep
```

bonedaddy, 2020-12-01:
That did it, thanks.

closed 2020-12-01 by bonedaddy

tobozo, 2020-12-05:
Followup on the root cause of this:

The library version mess is now fixed in platformio registry \o/

Due to a few errors in the library.properties files the last valid version was from June 2020, also for some reason I was locked out of my platformio profile and couldn't force any update, but the tech support solved that and the side effects are gone.

Platorfmio 5.x also solves many problems such as finding the proper SD.h library version or solving cross-dependencies more gracefully, so running "pio upgrade" and "pio update" is a good idea.