G4lile0/ESP32-WiFi-Hash-Monster #21: Better PCAP File Names?

bonedaddy, 2020-12-02:
Was playing around with the hash monster earlier, and noticed that the captured pcap files are named like "0000.pcap", "000A.pcap". I have a few questions, and suggestions.

# Question

* Files named with all numbers (0000.pcap, 0001.pcap, etc...) are empty files, were these failed captures?
* Files named with numbers and lettesr (000A.pcap, 000B.pcap, etc..) are filled with data, and these are the potentially crackable captures?

# Suggestion

I'm not sure if this is possible, since I haven't done a lot of reading of the codebase, but it might be nice to save the SSID name the capture corresponds to. For example with pwnagotchi each SSID has its own corresponding pcap file. Is this possible with the hash monster? If so I would be willing to take a stab at implementing it, however I would need a bit of guidance as to the proper part of the codebase to alter.

tobozo, 2020-12-02:
Without a RTC module, files are created with an inacurate timestamp, so the only chronological sorting that's left is by incrementing file names, zero-padded hexadecimal (%04x) seemed better at that job than decimal (%d).

I'm not sure why empty files are created though, but it may have something to do with channel hopping.

>  For example with pwnagotchi each SSID has its own corresponding pcap file

That sounds like naming files with (trusted) user input, how about having folders named after the mac address ?

bonedaddy, 2020-12-03:
> I'm not sure why empty files are created though, but it may have something to do with channel hopping.

That's probably the case, I'm using smart channel mode.

> That sounds like naming files with (trusted) user input, how about having folders named after the mac address ?

I should've clarified, it does SSID_MAC.pcap, but even just folders with mac address is great.

tobozo, 2020-12-03:
The easiest way I can think of to test that is to create a `Buffer::setFolder( const char* macAddressStr )` and update the contents of `Buffer::folderName` with the stringified mac address on every mac change.

Files in folders will still be named after an hex incremented number though, and maintaining per-folder proper numbering could be a hassle.

I don't recommend using SSIDs for naming the files without proper content-filtering and/or escaping (avoid path traversal), plus removing some chars from the SSID to get a clean file/folder name would eventually alter the very data you're trying to save.

Maybe saving the corresponding MAC/SSID pair in a separate file (as json, csv, whatever) whille keeping the pcap file in their respective mac folders would be safer? This would require an extra i/o strategy though, and possibly some content lookup to prevent redundancy.

bonedaddy, 2020-12-03:
Ah thats true. I think you are right the best idea is to save the corresponding MAC/SSID pair in a separate file.