G4lile0/ESP32-WiFi-Hash-Monster #22: Provided PCAP Conversion Instructions Dont Always Work

bonedaddy, 2020-12-03:
Not really sure if this is an applicable issue since it's not really related to the code, but reporting anyways. The instructions for converting the pcap files into hccapx files dont always work. I've been running some tests, and I'm not exactly sure why but running `cap2hccapx` sometimes results in writing 0 handshakes. However using `hcxpcapngtool` against the same pcap works.

For example here's output from my home wifi network running `cap2hccapx` against the captured traffic:

```
$> cap2hccapx nope.pcap capture.hccapx

Networks detected: 1

[*] BSSID=nope ESSID=nopenope (Length: 10)

Written 0 WPA Handshakes to: capture.hccapx
```

However if we try `hcxpcapngtool` against the pcap file, it works

```
$> hcxpcapngtool --all -o capture nope.pcap

summary capture file
--------------------
file name................................: nope.pcap
version (pcap/cap).......................: 2.4 (very basic format without any additional information)
timestamp minimum (GMT)..................: 02.12.2020 00:52:08
timestamp maximum (GMT)..................: 02.12.2020 00:52:08
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianess (capture system)...............: little endian
packets inside...........................: 4
frames with correct FCS..................: 4
BEACON (total)...........................: 2
EAPOL messages (total)...................: 2
EAPOL RSN messages.......................: 2
ESSID (total unique).....................: 1
EAPOL ANONCE error corrections (NC)......: not detected
EAPOL M1 messages........................: 2
PMKID (total)............................: 2
PMKID (best).............................: 2
PMKID written to combi hash file.........: 2

Warning: missing frames!
This dump file contains no important frames like
authentication, association or reassociation.
That makes it hard to recover the PSK.

Warning: missing frames!
This dump file contains no undirected proberequest frames.
An undirected proberequest may contain information about the PSK.
That makes it hard to recover the PSK.

$> > cat capture
WPA*01*nope*nope*nope*nope***
````

tobozo, 2020-12-03:
from [https://hashcat.net/wiki/doku.php?id=cracking_wpawpa2][hashcat.net wiki]:
> hcxpcapngtool support new WPA-PBKDF2-PMKID+EAPOL hash format

could it explain the difference ?

this hints at updating the ReadMe and quoting the hashcat.net wiki:

```
    Get hashcat-utils from https://github.com/hashcat/hashcat-utils
    Use cap2hccapx to convert locally

or

    Get hcxtools from https://github.com/ZerBea/hcxtools
    Use hcxpcapngtool to convert locally and/or to get the PMKID
```

bonedaddy, 2020-12-03:
I think that does indeed explain the difference. It might be worth having both instructions in the readme, or perhaps just using hcxpcapngtool as it should support the same format that cap2hccapx does.