G4lile0/ESP32-WiFi-Hash-Monster #23: Install onto SD Updater problems

tonysoprano187, 2021-01-12:
Hey, Im new to this and would like some advice in installing wifi hash monster onto my SD Updater instead of just burning the whole thing onto my M5Stack.... So i can use other applications. 

Many Thanks

tobozo, 2021-01-12:
@tonysoprano187 just modify this line in the .ino sketch :  

    checkSDUpdater( SD, MENU_BIN, 5000 ); 

And you'll get a 5 seconds delay to choose between loading the "menu.bin" default app or let the wifi hash monster go on with its duties.

tonysoprano187, 2021-01-12:
@tobozo thanks! .... I modified this and when I try to compile the sketch in arduino it brings up the error 'TFT_eSprite' does not name a type?  

The error message is quite long, but can post it

tobozo, 2021-01-12:
you may need to update your libraries from the Arduino IDE Library Manager, please update those:

- LovyanGFX
- ESP32-Chimera-Core
- M5Stack-SD-Updater

also make sure you've selected "M5Stack", "M5Fire" or "M5Core2" from the boards menu before compiling

tonysoprano187, 2021-01-13:
@tobozo Amazing, thank you ... got it working after a few tries.

closed 2021-01-13 by tonysoprano187