G4lile0/ESP32-WiFi-Hash-Monster #30: Buttons not working on M5StackCore2

hexsecs, 2021-09-28:
The A,B,C buttons don't appear to be working on M5Stack Core2.

tobozo, 2021-09-28:
Thanks for reporting that, the reason Buttons are not working is that it wasn't intented to work on this device (M5Core2 support was only a proof of concept and M5Core Buttons use `digitalRead()` ).

But since a a quick and dirty implementation exists, I'm pushing the *Physical Buttons Emulation for M5Core2 <=> M5Stack* into ESP32-Chimera-Core.

Changes will soon be available as a library update in both platformio and arduino library manager.

If you're in a hurry you can still check out the [1.2.0 branch](https://github.com/tobozo/ESP32-Chimera-Core/tree/1.2.0).

Please note this is a partial implementation, e.g. long pushes will probably be registered as short pushes, so don't expect 100% compatibility with the behaviour you get on a M5Fire.

tobozo, 2021-10-01:
bump

I've been adjusting the ESP32-Chimera-Core library for M5Core2, short and long presses are now responding properly on my local build.

I've also been testing support for the M5Core2's BM8563 RTC module, and ESP32-Chimera-Core will be implicitely syncing RTC time to system time in the next release.

Now M5Core2 implementation becomes more interesting than M5Fire implementation :-)

[image]

I've also made a few changes to the application UI, fixed some issues with sprites overlapping, but it needs some refactoring before being published, I'll update this post when the code is available.

[image]