G4lile0/ESP32-WiFi-Hash-Monster #34: Undirected PROBE REQUEST and EAP REQUEST/RESPONSE ID frames missing

xianclasen, 2021-11-20:
It seems packets may be filtered before they are written to disk and exclude frames that tools like Hashcat want to see when converting to useful file formats for cracking. This results in errors from hcxpcapngtools like this one:

```
Warning: missing frames!
This dump file contains no important frames like
authentication, association or reassociation.
That makes it hard to recover the PSK.

Warning: missing frames!
This dump file contains no undirected proberequest frames.
An undirected proberequest may contain information about the PSK.
That makes it hard to recover the PSK.
```

I've not been able to get hcxpcaptools to accept the saved pcap files from Hash Monster due to this missing data.

allmeatdies, 2021-11-25:
I have had this same issue for quite some time. Additionally, while several .pcap files are written to disk, all of them are empty except for 0.pcap.

In my most recent test, running a freshly compiled installation for an hour generated 142 .pcap files - from 0.pcap to 141.pcap. 1-141 show a file size of 0 bytes.

Running hcxpcapngtool --all -o capture 0.pcap gives the following:

```

pcapngtool --all -o capture 0.pcap
hcxpcapngtool 6.2.4-96-gd8f56be reading from 0.pcap...

summary capture file
--------------------
file name................................: 0.pcap
version (pcap/cap).......................: 2.4 (very basic format without any additional information)
timestamp minimum (GMT)..................: 31.12.1969 18:00:00
timestamp maximum (GMT)..................: 31.12.1969 18:00:00
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11 (105) very basic format without any additional information about the quality
endianess (capture system)...............: little endian
EAPOL ANONCE error corrections (NC)......: not detected

Information: no hashes written to hash files

Warning: missing frames!
This dump file does not contain undirected proberequest frames.
An undirected proberequest may contain information about the PSK.
It always happens if the capture file was cleaned or
it could happen if filter options are used during capturing.
That makes it hard to recover the PSK.

Warning: missing frames!
This dump file does not contain important frames like
authentication, association or reassociation.
It always happens if the capture file was cleaned or
it could happen if filter options are used during capturing.
That makes it hard to recover the PSK.

Warning: missing frames!
This dump file does not contain enough EAPOL M1 frames.
It always happens if the capture file was cleaned or
it could happen if filter options are used during capturing.
That makes it impossible to calculate nonce-error-correction values.

session summary
---------------
processed cap files...................: 1
```

tobozo, 2021-11-25:
I recently read something like "don't echo to serial while in promisc mode" and this app seems to do exactly that.

Are there any improvement when all `Serial.print**` statements are commented out?

xianclasen, 2021-11-25:
I think this may be a part of a larger problem that the tool-chain in use by Hash Monster is a bit outdated. Aside from missing these packets, it outputs in pcap format (most tools now use pcapng), and hash modes 2500/2501, which have been deprecated for better formats for cracking (https://hashcat.net/wiki/doku.php?id=cracking_wpawpa2).

If it can be updated to capture the packets mentioned in the title, in pcapng format, it would go a long way.

tobozo, 2021-11-28:
Oh this mean the WiFi Hash Monster needs a feature upgrade rather than a bugfix?

Compiling with the new espressif32-sdk (2.0.x) throws a few deprecation warnings, so there's a small technical debt to satisfy first.

After that [`wifi_sniffer_packet_type2str`](https://github.com/G4lile0/ESP32-WiFi-Hash-Monster/blob/master/ESP32-WiFi-Hash-Monster/ESP32-WiFi-Hash-Monster.ino#L694-L702) will need to be extended, and the logic of [`wifi_promiscuous`](https://github.com/G4lile0/ESP32-WiFi-Hash-Monster/blob/master/ESP32-WiFi-Hash-Monster/ESP32-WiFi-Hash-Monster.ino#L717-L718) should be rewritten e.g. to remove FastLed and Serial writes from the callback function.

However that's as far as I can speculate for the roadmap. 
Can you point me out to a simple example using espressif sdk where these packets are captured?

xianclasen, 2021-11-29:
>Oh this mean the WiFi Hash Monster needs a feature upgrade rather than a bugfix?

I think this is correct. I am able to get pcap files and convert them to 2500/2501 hash files (using the old hcxpcaptool), but this is an old way of going about cracking WPA2.

>Can you point me out to a simple example using espressif sdk where these packets are captured?

This is getting outside of my knowledge of the subject, unfortunately. I believe that pwnagotchi have already implemented this and capture all 802.11 packets in pcapng without filtering or cleaning (which is what hcxpcapng prefers). I don't know if having a look at their codebase would help you.

Thank a ton for responding, btw.