G4lile0/ESP32-WiFi-Hash-Monster #38: SD Card shuts off

watdaf, 2022-04-01:
When performing scans, SD card turns off and needs to be manually enabled . Because of this, no hashes are being stored on the SD card

tobozo, 2022-04-02:
hey thanks for your feedback

Please provide more information on your configuration, system, platform, package version, device, etc.

FYI the SD Card is supposed to turn itself off automatically when too many write errors occur, this is a by design behaviour, e.g. to detect SD eject/insert events. 

Read errors car still occur though, when the SD card is too slow and the WiFi traffic heavy, or if the pcap folder is loaded with too many files.

Also since arduino-esp32-2.0.0 new read timeout symptoms appeared and SD remounting is broken.

The latest arduino-esp32 2.0.3-RC1 package has a fix for the initial SD mounting problem, but it's only available in the Arduino IDE and probably won't show up in platformio as it's only a release candidate.

watdaf, 2022-04-03:
Using the M5stack ESP32 Audrinio 1.8.19 

have tried several SD cards of multiple speeds/sizes

I can get a pcap file but it only contains the endpoints, no handshakes are ever captured/written

watdaf, 2022-04-03:
I was using the M5stack board so I switched to the ESP32 board (M5 Stack) v 1.0.6 to see if that makes a change

teltech84, 2022-10-13:
Same thing is occuring to me on M5StackCore2.   Any ideas on the fix?  As soon as the SD card is turned off I can't turn it back on again as the longpress is unresponsive.

tobozo, 2022-10-15:
This may be helpful to identify a better SD Card model.

https://github.com/yamamaya/M5-SDInfoViewer