G4lile0/ESP32-WiFi-Hash-Monster #39: How to configure correct pcap creation timestamp on sd card?

maxkarelov, 2022-05-16:
It's not clear for me how I can configure hash monster firmware to use correct timestamp of created file on sd card. Looks like RTC module works well with right time setup but dumped files have wrong date.

[image]

tobozo, 2022-05-16:
This doesn't sound like a normal behaviour, builtin RTC support for M5core2 was [added to Chimera-Core](https://github.com/tobozo/ESP32-Chimera-Core/blob/master/src/ESP32-Chimera-Core.cpp#L313) long ago and should be silently synched when M5.begin() is called.

One thing you can try is edit the lib_deps section of platformio.ini:

```ini
  LovyanGFX@0.4.3
  ESP32-Chimera-Core@1.2.4
  M5Stack-SD-Updater@1.1.3
```

Removing the version numbers should be enough of a test:

```ini
  LovyanGFX
  ESP32-Chimera-Core
  M5Stack-SD-Updater
```

Also make sure your pcap folder is empty before testing anything, your screenshot shows the "Date modified" byt not the "Date created", and the SD driver has a long history of bugs :-)

hexsecs, 2022-05-19:
I'm experiencing the same issue on Core2. Tried bumping packages to latest as @tobozo suggested, same issue. My RTC is set correctly (I have a quick and dirty program to set it using NTP). 

[image]

Here are my dependency versions...

```
Dependency Graph
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
|-- FS @ 2.0.0
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- HTTPClient @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- WiFi @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- Update @ 2.0.0
|-- Preferences @ 2.0.0
|-- LovyanGFX @ 0.4.17
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- ESP32-Chimera-Core @ 1.4.2
|   |-- LovyanGFX @ 0.4.17
|   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- SD @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- SD_MMC @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- Preferences @ 2.0.0
|-- M5Stack-SD-Updater @ 1.2.0
|   |-- FS @ 2.0.0
|   |-- LittleFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- SD @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- SD_MMC @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- SPIFFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- Preferences @ 2.0.0
|-- SD @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- SPI @ 2.0.0
```

tobozo, 2022-05-22:
I know it's annoying to read that so I apologize in advance, but it worked for me using Arduino and the latest stable package.

[image]

[image]

I couldn't reproduce with platformio either.

The latest espressif [changelog](https://github.com/espressif/arduino-esp32/discussions/6684), highlights a long history of problems with the SD driver, but comes with fixes for SDl and I believe this will solve the issue you're describing.

It isn't clearly documented how to use 2.0.3 framework package with 4.x espressif32 platform though, so here's my platformio.ini:

```ini

[platformio]
src_dir = ESP32-WiFi-Hash-Monster
;default_envs = m5stack-fire
;default_envs = m5stack-core-esp32
default_envs = m5stack-core2
;default_envs = odroid_esp32

[env]
platform = espressif32@^4
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/archive/refs/tags/2.0.3.zip
framework = arduino
upload_speed = 921600
monitor_speed = 115200
; deep needed in order to pick up SD.h dependency of M5Stack-SD-Updater
lib_ldf_mode = deep
lib_compat_mode = strict
; lib_extra_dirs = ~/Documents/Arduino/libraries/
lib_deps =
  LovyanGFX@^0.4.0
  ESP32-Chimera-Core@^1.4.0
  M5Stack-SD-Updater@^1.2.0

[env:m5stack-fire]
board = m5stack-fire
lib_deps =
  ${env.lib_deps}
  FastLED@3.4.0

[env:m5stack-core-esp32]
board = m5stack-core-esp32
lib_deps =
  ${env.lib_deps}
lib_ignore =
  M5Stack

[env:m5stack-core2]
board = m5stack-core2
lib_deps =
  ${env.lib_deps}
lib_ignore =
  M5Core2

[env:odroid_esp32]
board = odroid_esp32
lib_deps =
  ${env.lib_deps}

```

please let me know how it went so I can eventually submit a PR for this

tobozo, 2022-10-15:
This may be helpful to identify a better SD Card model.

https://github.com/yamamaya/M5-SDInfoViewer