G4lile0/ESP32-WiFi-Hash-Monster #49: Got it but...

Koekebeest, 2023-06-12:
Hello world,

Finally got the hash monster running on my m5stack core 2 but the touchscreen buttons don't seem to work properly. Any ideas on why that might be and on how to fix this? I'm also curious about getting a deauth'er on the core2 since that would be a handy tool for the hashmonster to integrate with. Is there an integrated version of this somewhere already? Or do I need a seperate module for esp2866 for this?

Thanks for creating this I'm having super much fun with this already.

zerospidey, 2023-07-09:
Would you be able to share the setup you used? I've been trying to make similar edits but have been unsuccessful, Thanks in advance!

shanejohn15, 2023-07-10:
Second that? I can't get my m5stack to get this, or marauder, I can get the no screen version of marauder on my m5stack core 2 but not hash monster, I've changed them to LGFX, but do you change them in the first and second time in each line or just the first time? And NVS TO spi I didn't put them ()

Dviros, 2023-07-12:
Third that, PCAPs are being saved to the SD card as empty files

Koekebeest, 2023-07-20:
Sorry for the late reply. I used Arduino Cloud and that seemed to help a lot because most (if not all)  libs for this build are already there. 
This is the sketch I am using now verified and uploaded via Arduino Cloud Creator. 

[M5Stack_SD_Menu_old_libs.zip](https://github.com/G4lile0/ESP32-WiFi-Hash-Monster/files/12112686/M5Stack_SD_Menu_old_libs.zip)

Hope it helps. Cheers

Koekebeest, 2023-07-20:
> Third that, PCAPs are being saved to the SD card as empty files

That's apparently a more common "issue" it's all about the pcaps with the data. Should be one or two there.

Koekebeest, 2023-07-20:
> Second that? I can't get my m5stack to get this, or marauder, I can get the no-screen version of marauder on my m5stack core 2 but not hash monster, I've changed them to LGFX, but do you change them in the first and second times in each line or just the first time? And NVS TO spi I didn't put them ()

Haven't tried Marauder yet. Focussing back on getting the SD-stack-updater working so I can put more than one application on it like the Marauder. No luck yet but haven't tried the latest updates from Tobozo yet. Anyway, hope the sketch works for you guys.

Koekebeest, 2023-07-20:
> Would you be able to share the setup you used? I've been trying to make similar edits but have been unsuccessful, Thanks in advance!

Hope it helps!

chetlaory, 2023-07-28:
> > Would you be able to share the setup you used? I've been trying to make similar edits but have been unsuccessful, Thanks in advance!
> 
> Hope it helps!

I made an account just to say this absolutely helps. Using the provided sketch, specifically on Arduino cloud - I was able to get this running on the core2. Thank you so much! 🍡

Koekebeest, 2023-07-31:
> I made an account just to say this absolutely helps. Using the provided sketch, specifically on Arduino cloud - I was able to get this running on the core2. Thank you so much! 🍡

Glad it worked out! You are most welcome

CaesarAtmos, 2023-09-07:
> > Would you be able to share the setup you used? I've been trying to make similar edits but have been unsuccessful, Thanks in advance!
> 
> Hope it helps!

Thank you so much for this sketch file! I had to use the cloud editor with your sketch but after that it uploaded perfectly.

Koekebeest, 2023-09-08:
> > > Would you be able to share the setup you used? I've been trying to make similar edits but have been unsuccessful, Thanks in advance!
> > 
> > 
> > Hope it helps!
> 
> Thank you so much for this sketch file! I had to use the cloud editor with your sketch but after that it uploaded perfectly.

You are most welcome :)

samad350z, 2023-12-03:
> Sorry for the late reply. I used Arduino Cloud and that seemed to help a lot because most (if not all)  libs for this build are already there. 
> 
> This is the sketch I am using now verified and uploaded via Arduino Cloud Creator. 
> 
> 
> 
> Woops wrong sketch. Sorry about that....let me take a look again. Here it is: 
> 
> [ESP32-WiFi-Hash-Monster.zip](https://github.com/G4lile0/ESP32-WiFi-Hash-Monster/files/12121909/ESP32-WiFi-Hash-Monster.zip)
> 
> 
> 
> 
> 
> Hope it helps. Cheers
> 
> 

You are awesome! Thanks!