G4lile0/ESP32-WiFi-Hash-Monster #51: 'Core2' Buttons Fix?

417bay, 2023-07-29:
Hi, Do the buttons on Core2 work properly? I'm not able to see why they don't work for me. Is it a known issue? I'm using Visual Studio Code and PlatformIO, and I'm cloning this repo. I can flash it without issues, but I can't use the buttons. What am I doing wrong?

417bay, 2023-07-30:
Ok i fixed that...

rustyorb, 2023-08-01:
I am having this issue on the AWS version that does not have tactile buttons, but touch buttons..

pedroCX486, 2023-08-20:
> Ok i fixed that...

Help everyone and post how you did it.

Cheers

417bay, 2023-08-25:
Oh, my apologies. 
I initially thought I couldn't find a solution.I thought that the problem is obvious (This is only my 2nd Arduino project.) 

It appears there's an issue with the library in that repository ( they are 2 years old). I've been using Visual Studio out of habit, which was a mistake. 
That is how I flash it.

Flash Your Core2 to Factory Settings using M5Burner
Visit Arduino Cloud and create a free account.
Download the desktop client so you can connect your device. (it took me a while to realise)
Surprisingly, they have all the required libraries, just copy the source code.
After successfully flashing, unplug your device and wait approximately 60 seconds. After that, insert the SD Card into your Core2 and power it on. Your buttons should function as expected.

If, for some reason, your device doesn't power on, plug it back into the USB and wait. This has occurred for me two or three times.

zloneotm, 2024-01-30:
Currently waiting if they update the file for core2. 
Not able to boot correctly. I will follow up with the steps listed above. 
Platformio didn't have all the libraries for me nor did arduino.

417bay, 2024-02-02:
> Currently waiting if they update the file for core2. Not able to boot correctly. I will follow up with the steps listed above. Platformio didn't have all the libraries for me nor did arduino.

Just use free Adurion Cloud here https://cloud.arduino.cc/
Install Agent
Use https://app.arduino.cc/ to flash

zloneotm, 2024-02-04:
> > Currently waiting if they update the file for core2. Not able to boot correctly. I will follow up with the steps listed above. Platformio didn't have all the libraries for me nor did arduino.
> 
> 
> 
> Just use free Adurion Cloud here https://cloud.arduino.cc/
> 
> Install Agent
> 
> Use https://app.arduino.cc/ to flash
> 
> 
> 
> [ESP32-WiFi-Hash-Monster.zip](https://github.com/G4lile0/ESP32-WiFi-Hash-Monster/files/14140518/ESP32-WiFi-Hash-Monster.zip)
> 
> You can use my sketch

Still not working... I believe my core2 is a newer version. 
Error: " rat:ox1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2) ) waiting for download " 
While trying to boot. Thank you though! 
Going to stick with the pwnagotchi...