G4lile0/ESP32-WiFi-Hash-Monster #58: M5StackCore2 Compile

BrunoKlu, 2023-11-26:
Hello,

Anyone having issue compiling the latest code for M5StackCore2 ?

Here is the compile output : 

```
`/usr/local/bin/arduino-cli compile --fqbn esp32:esp32:m5stack-core2:CPUFreq=240,DebugLevel=none,EraseFlash=none,PSRAM=enabled,PartitionScheme=default,UploadSpeed=921600 --build-cache-path /tmp --output-dir /tmp/1924142457/build --build-path /tmp/arduino-build-F5CBD01E662C95E45A043D910EF68964 --library /mnt/create-efs/webide/86/6b/866b96c9b47258e3fc1e3b4c079ee707:nobsklu/libraries_v2/EEPROM --library /mnt/create-efs/webide/86/6b/866b96c9b47258e3fc1e3b4c079ee707:nobsklu/libraries_v2/ESP32-Chimera-Core --library /mnt/create-efs/webide/86/6b/866b96c9b47258e3fc1e3b4c079ee707:nobsklu/libraries_v2/HID --library /mnt/create-efs/webide/86/6b/866b96c9b47258e3fc1e3b4c079ee707:nobsklu/libraries_v2/SPI --library /mnt/create-efs/webide/86/6b/866b96c9b47258e3fc1e3b4c079ee707:nobsklu/libraries_v2/SoftwareSerial --library /mnt/create-efs/webide/86/6b/866b96c9b47258e3fc1e3b4c079ee707:nobsklu/libraries_v2/Wire /tmp/1924142457/ESP32-WiFi-Hash-Monster

/mnt/create-efs/webide/86/6b/866b96c9b47258e3fc1e3b4c079ee707:nobsklu/libraries_v2/Wire/src/utility/twi.c:26:10: fatal error: avr/io.h: No such file or directory

#include <avr/io.h>`
```

If someone have a idea here ;)

Thanks a lot

limes-github, 2024-01-12:
Look here: https://github.com/G4lile0/ESP32-WiFi-Hash-Monster/issues/61

BrunoKlu, 2024-01-21:
Thanks will try

closed 2024-01-21 by BrunoKlu