G4lile0/ESP32-WiFi-Hash-Monster #60: Installing on M5Go 2.7

Gottdestodes, 2023-12-18:
I am having a hard time getting this to install on the M5Go 2.7. Will this work on the M5Go, if so what is the proper procedure to get it working. I have tried with Arduino IDE but I never seem to get that program to work. I tried with platformIO and tried it as the Fire. It installed ok but constantly kept saying checking SD card. Sd card is installed and formatted as fate2. I have also tried using the other options for install but all fail. 

Can someone please help me out

vjtod, 2024-03-29:
I also purchased the M5Stack M5GO IoT Starter Kit V2.7. Will this work?

vjtod, 2024-03-31:
> @Gottdestodes <https://github.com/Gottdestodes> @vjtod
> I was able to install on Fire 2.7 using Platform in VSCode. In
> Platformio.ini, comment "default_envs = m5stack-fire" and uncomment
> "default_envs = m5stack-core-esp32". Also, I changed the PlatformIO Project
> Environment to "m5stack_core_esp32". You won't have the M5Fire led effects
> but everything works as expected. Hope that helps.

Thank you very very much For your help and taking the time to post your
experience. This has been a learning experience and so much fun.

hostmadns, 2024-03-31:
@vjtod - If you need help with anything else, feel free to ask :-)

vjtod, 2024-03-31:
Absolutely, thank you again.