G4lile0/ESP32-WiFi-Hash-Monster #61: Compilation on M5Stack Core2 - working buttons.

limes-github, 2024-01-07:
Compilation and upload worked flawlessly on Arduino IDE with functional buttons.
Insert SD card, connect M5Stack Core2 by USB and upload. Use versions of board and library below, worked for me.

Boards manager: M5Stack version 2.0.0

Library:
M5Stack-SD-Updater version 1.2.8
M5Core2 version 0.1.8
LovyanGFX version 1.1.12
ESP32-Chimera-Core version 1.5.2
Module_GRBL_13.2 version 0.0.3

I hope it is useful for anyone.

caz1121, 2024-01-12:
Thank you very much... This is the first time in 4 days of trying it has booted into Hashmonster!! 

But once turned off after running, the screen stays black and will not turn back on for a second boot. Blowing my mind.

I had to roll the Boards Manager to version 2.0.0 and it did exactly as you said.
Now to understand why it won't boot again, has anyone experienced the same issue?

limes-github, 2024-01-12:
> Thank you very much... This is the first time in 4 days of trying it has booted into Hashmonster!!
>
> But once turned off after running, the screen stays black and will not turn back on for a second [...]

I am glad you got it running, never experienced the issue you mentioned though. Maybe someone with greater knowledge can help here.

caz1121, 2024-01-12:
> I am glad you got it running, never experienced the issue you mentioned though. Maybe someone with greater knowledge can help here.

I have re-compiled and uploaded and it now boots flawlessly. 

You are definitely on my Christmas card list!!

limes-github, 2024-01-12:
> I have re-compiled and uploaded and it now boots flawlessly.
>
> You are definitely on my Christmas card list!!

Damn, need to wait another year ;) 
But seriously - I also tried almost everything on this, spent a lot of time - that is the reason why I shared this. So save the hassle.

BrunoKlu, 2024-01-21:
Seems to be working for me also ! My only problem is that I cannot find an SD card that is mounting on my M5stackCore2 but that's another story. Thanks a lot for your help

zloneotm, 2024-01-30:
Thank you for the tips!! 
Still having issues on my Core2 however.
Compiling seems to succeed however I'm having booting issues, black screen. 
I'm on board manager M5stack 2.0.0 and followed the library version's accordingly. Do I need the supporting dependencies as well? 
Library is much bigger than the one listed above. 
Also what are the flashing configurations? 
I also have the esp tool programmer. Any help is appreciated!!! When I get home, I will list the error listed in serial monitor when booting. 
Something ending in "waiting for download".