G4lile0/ESP32-WiFi-Hash-Monster #62: Issues with PSRAM and M5Stack Fire v2.7

toto99303, 2024-01-16:
Hello,

Can anyone verify Wifi Hash Monster is working with M5Stack Fire v2.7 ?
I just received my M5 Fire v2.7 and I have strange problems after uploading:

The M5 Fire boot loops and at some point it shows:

`E (934) spiram: SPI SRAM memory test fail. 95419/131072 writes failed, first @ 3F80ACC0

[   297][E][esp32-hal-psram.c:71] psramInit(): PSRAM test failed!
ESP32-Chimera-Core initializing...OK
[   569][E][M5StackUpdater.h:290] checkSDUpdater(): Booting with reset reason: 12
[  8707][E][Buffer.cpp:11] init(): Panic! Can't malloc 24576 bytes for buffer 1
Error, not enough memory for buffer

E (871) spiram: SPI SRAM memory test fail. 21/131072 writes failed, first @ 3F80BD20

[   297][E][esp32-hal-psram.c:71] psramInit(): PSRAM test failed!
ESP32-Chimera-Core initializing [Board=M5STACK_FIRE] [Variant=m5stack_fire] 
ESP32-Chimera-Core started
[   562][E][M5StackUpdater.hpp:548] checkSDUpdater(): Booting with reset reason: 1
[   745][W][ESP32-WiFi-Hash-Monster.ino:466] initSpritesTask(): Leaving initSprites task !
[  6756][E][Buffer.cpp:11] init(): Panic! Can't malloc 24576 bytes for buffer 1
Error, not enough memory for buffer`

Strangely when uploading their own UFlow_Fire firmware via M5Burner everything seems fine, not sure how to test PSRAM with Uflow, just tested some small demos....
So could this be a hardware PSRAM issue or some kind of code problem with Fire v2.7 ?

Thanks!

toto99303, 2024-01-17:
I tested the available PSRAM with this sketch:

#include <M5Stack.h>

void setup() {
  M5.begin();
  Serial.begin(115200);
  
  // Initialize PSRAM
  if(!psramInit()) {
    Serial.println("Failed to initialize PSRAM");
    while(1);
  }

  // Test PSRAM
  testPSRAM();
}

void loop() {
  // Your main code here
}

void testPSRAM() {
  size_t bufferSize = 1024;  //1KB
  uint8_t* buffer = (uint8_t*)ps_malloc(bufferSize);

  for (int i=3800; i<4097; i++){ //test the max possible size starting from 3800 KB
      bufferSize = i * 1024;  //KB
      buffer = (uint8_t*)ps_malloc(bufferSize);
      if (buffer == NULL) {
        Serial.println("Failed to allocate " + String(i) + " KB from PSRAM, will continue with " +String(i-1) + " KB");
        free(buffer);
        bufferSize = (i-1) * 1024;
        buffer = (uint8_t*)ps_malloc(bufferSize);
        break; //continue with the max possible allocated PSRAM
      }
      delay(10);
      Serial.println("Successfully allocated: " + String(i) + " KB");
      free(buffer);
  }

  // Write data to PSRAM
  for (size_t i = 0; i < bufferSize; i++) {
    buffer[i] = i % 256;
  }

  // Read and verify data from PSRAM
  for (size_t i = 0; i < bufferSize; i++) {
    if (buffer[i] != (i % 256)) {
      Serial.println("PSRAM test failed");
      while(1);
    }
  }

  Serial.println("PSRAM test passed");

  // Free allocated memory
  free(buffer);
}

Which returns this:

Successfully allocated: 4026 KB
Successfully allocated: 4027 KB
Successfully allocated: 4028 KB
Successfully allocated: 4029 KB
Successfully allocated: 4030 KB
Successfully allocated: 4031 KB
Failed to allocate 4032 KB from PSRAM, will continue with 4031 KB
PSRAM test passed

Can anyone test this sketch on their Fire devices and confirm the same output?

I'm still confused if PSRAM is the issue here....

udodill, 2024-03-05:
hello @toto99303,
I tested the sketch and it works on my M5Stack-Fire.

[TestRam | Arduino IDE 2.3.2_001.pdf](https://github.com/G4lile0/ESP32-WiFi-Hash-Monster/files/14500552/TestRam.Arduino.IDE.2.3.2_001.pdf)