G4lile0/ESP32-WiFi-Hash-Monster #63: M5Stack Core 2

zloneotm, 2024-01-31:
Board Manager: M5Stack 2.0.0
Library: 
ESP32-Chimera-core=1.5.2
LovyanGFX=1.1.12
M5Core2=0.1.8
M5Stack-SD-Updater=1.2.8
Module_GRBL_13.2=0.0.3

After flashing I am getting this while trying to boot:

rst: 0x1 (POWERON_RESET), boot: 0x3 (DOWNLOAD BOOT (UARIO/UARTI/SDIO_REI_REO_V2)) waiting for download

Can anyone help ?

zloneotm, 2024-01-31:
I believe I need help configuring my library if anyone can share theirs please!

vassalo, 2024-02-21:
I'm also trying to run the project on my Core2 but it doesn't give any sign of life... I'm using VS Code with platformio to build and upload the project to the device.

ehalexander2, 2024-03-11:
Same problem. After successful upload, I get black screen and no signs of life.

evilgenius79, 2024-05-23:
> Board Manager: M5Stack 2.0.0 Library: ESP32-Chimera-core=1.5.2 LovyanGFX=1.1.12 M5Core2=0.1.8 M5Stack-SD-Updater=1.2.8 Module_GRBL_13.2=0.0.3
>
> After flashing I am getting this while trying to boot: [...]

can you help with this? i have tried older versions as well. google no good either.

`c:\Users\Shop\Documents\Arduino\libraries\ESP32-Chimera-Core\src\ESP32-Chimera-Core.cpp: In member function 'void ChimeraCore::ECCKernel::begin(bool, bool, bool, bool, bool)':
c:\Users\Shop\Documents\Arduino\libraries\ESP32-Chimera-Core\src\ESP32-Chimera-Core.cpp:53:11: error: 'class AXP192' has no member named 'begin'
       Axp.begin();
           ^~~~~

exit status 1

Compilation error: exit status 1`