G4lile0/ESP32-WiFi-Hash-Monster #65: M5 Stack Core 2 Compilation Issue

BeowolfK, 2024-04-29:
Hello,
I want to burn Hash monster on a new M5 Core 2 but when I try to compile on Arduino IDE I have the following error : 

```cpp
/home/kenan/Arduino/libraries/ESP32-Chimera-Core/src/ESP32-Chimera-Core.cpp: In member function 'void ChimeraCore::ECCKernel::begin(bool, bool, bool, bool, bool)':
/home/kenan/Arduino/libraries/ESP32-Chimera-Core/src/ESP32-Chimera-Core.cpp:53:11: error: 'class AXP192' has no member named 'begin'
       Axp.begin();
           ^~~~~

exit status 1

Compilation error: exit status 1
```

I' ve installed Chimera Core and SD Updater with the Library manager
I've changed nothing in the code, just download it and trying to compile with Arduino IDE
If someone can help me
Thanks

hostmadns, 2024-04-29:
Have you tried using PlatformIO in VSCode?

BeowolfK, 2024-04-30:
Hello,
Yes that's work with PlatformIO, I didn't know this tools before
Thank you :)

closed 2024-04-30 by BeowolfK